orangepi-xunlong/wiringOP #4: orange pi i96 support?

supersh86, 2019-10-03:
How to install on i96?

All available boards:
   0. OrangePi_PC2
   1. OrangePi_A64
   2. OrangePi_ZERO
   3. OrangePi_H3
   4. OrangePi_LITE2
   5. OrangePi_H3_ZEROPLUS2
   6. OrangePi_3
   7. OrangePi_RK3399
   8. OrangePi_ONEPLUS
Choice:

phtdacosta, 2021-03-08:
I want to second your demand as I also bought a **i96 and a 2G-IOT** and both lack support from this library. I like Orange Pi boards but that lack of support throughout the years is, _being really polite here_, annoying to say at least.

Chandler-Kluser, 2021-12-10:
I would like to have the support too!! I have one i96 here

alexgubanow, 2022-01-17:
Hi @supersh86 and @Chandler-Kluser  unfortunately it is not implemented. I could try to look if it is hard to do, can you highlight sweet spot I can start with?
Hi @phtdacosta 2g-iot is at least mentioned in code, try compile with orangepi2giot as board name. Also, can you describe your issues with 2g-iot board?

Chandler-Kluser, 2022-01-18:
@alexgubanow I will take a look at the code and start to learn about it, if there is another way to help I will be glad to do it.

henriclinton, 2023-04-07:
where I can get support about the [orange pi](https://electronicshacks.com/boards/orange-pi/)

baiywt, 2025-03-31:
For 2G-IoT and I96, please use this: https://github.com/OrangePiLibra/WiringPi.

closed 2025-04-01 by orangepi-xunlong