orangepi-xunlong/wiringOP #9: Orange Pi One Plus wiringPiISR not working

RRcom, 2020-02-25:
When using c++ wiringPiISR()  I get this error (even using root)

> gpio: Unable to open GPIO edge interface for pin 0: Permission denied
wiringPiISR: unable to open /sys/class/gpio/gpio229/value: No such file or directory

here is my code

#include <wiringPi.h>

int main()
{
        wiringPiSetup();

        wiringPiISR(1, INT_EDGE_FALLING, []() {
        });

        return 0;
}

lsafelix75, 2020-05-19:
I have the same problem for lite 2

lsafelix75, 2020-05-20:
@RRcom you need to do some hacking on the gpio/gpio.c and look for this function and comment out the BOLD code. The original wirinpi function is expecting bcmGpio as input pin and here the code reconvert again (probably it assumes input pin is wiringPi pin). Remove it then you will be fine but i am not sure what is the implication for others. May be the author knows it. 

```
void doEdge (int argc, char *argv [])
{
  FILE *fd ;
  int pin ;
  char *mode ;
  char fName [128] ;

  if (argc != 4)
  {
    fprintf (stderr, "Usage: %s edge pin mode\n", argv [0]) ;
    exit (1) ;
  }

  pin  = atoi (argv [2]) ;
#ifdef CONFIG_ORANGEPI
 **# pin = pinToGpioOrangePi[pin];**
#ifdef CONFIG_ORANGEPI_RK3399
	pin += 1000;
#endif

#endif
......
}
```

closed 2020-11-10 by orangepi-xunlong