orangepi-xunlong/wiringOP #16: Orange Pi Zero Plus and relay functionality issue:

madurani, 2020-04-20:
Hello,

I have problem with "Orange Pi Zero Plus" and functionality of Relay module (electrical circuit= normally closed).
Relay module is attached on pin/port number "0".
When I set up pin number "0" from "OFF" to "OUT" state with command:
```
gpio export 0 out
```
state is changed from OFF to OUT and parameter in column "V"  is "0". It should mean that relay stays turned of.

```
+------+-----+----------+------+---+
| GPIO | wPi |   Name   | Mode | V |
+------+-----+----------+------+---+
|   12 |   0 |    SDA.0 |  OUT | 0 |
```
Problem is, that relay immediately after execution of mentioned command enables(turns on) output electrical circuit. Relay module is behaving like it was turned on, however it is not supposed to be (since 0= turned off). 

I can change this inconvenient situation with command for turning on (1=turned on) and output is then turned off.  These commands work, but they work in reversed way and in result everything works reversed. 

`gpio write 0 1`
```
gpio write 0 0 >>> should turn off output circuit, for me it turns circuit ON

gpio write 0 1 >>> should turn on output circuit, for me it turns circuit OFF
```
I hope this was clear, it is hard to describe what is happening. 

Can anyone please advise me on what is going wrong and how to fix my issue?

Thank you

orangepi-xunlong, 2020-04-21:
Hi,

We have confirmed that the output voltage from this operation is correct.

It is possible that the output voltage is correct, but the relay behaves in the opposite way to the output voltage.

Can you measure the voltage of this pin? This will confirm that the output is correct or not.

Thank you!

madurani, 2020-04-21:
Hello,

Measurations were done. 
I use two pins 0 and 1. Voltage on input is 5.04V. Below are the values of all states of pins:
State OFF, on both pins is 0V.
State OUT, on both pins is write value 0 and measure Voltage is 0.13V
State OUT, on both pins is write value 1 and measure Voltage is 3.35V

It looks like an output electricity circle is connected to terminal "normally close'', but it is on terminal "normally open", because when OragePI is turned off, electricity circle is interrupted.

Relay module which i use:
https://www.aliexpress.com/item/32804288816.html?spm=a2g0s.9042311.0.0.27424c4dh1jIeH

Thank you.

madurani, 2020-05-02:
Problem was fixed. I found out, that exist relays with low and high level trigger. My relay was with low trigger and it was a reason of reverse functionality. I replaced relay and now all work correctly.

closed 2020-05-02 by madurani