orangepi-xunlong/wiringOP #22: Can't identify OPi Lite properly

inactive-virus, 2020-06-20:
Hardware: Orange Pi Lite
OS: Armbian Buster/Focal
OPi Lite has an H3+ CPU and 40-Pin GPIO, however native-compiled wiringOP treated it as H6 and show a 26-Pin layout when using `gpio readall`. Is that a bug?
```
 +------+-----+----------+------+---+  OPi H6  +---+------+----------+-----+------+
 | GPIO | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | GPIO |
 +------+-----+----------+------+---+----++----+---+------+----------+-----+------+
 |      |     |     3.3V |      |   |  1 || 2  |   |      | 5V       |     |      |
 |  230 |   0 |    SDA.1 |   IN | 0 |  3 || 4  |   |      | 5V       |     |      |
 |  229 |   1 |    SCL.1 |   IN | 0 |  5 || 6  |   |      | GND      |     |      |
 |  228 |   2 |     PWM1 |   IN | 0 |  7 || 8  | 0 | IN   | PD21     | 3   | 117  |
 |      |     |      GND |      |   |  9 || 10 | 0 | IN   | PD22     | 4   | 118  |
 |  120 |   5 |    RXD.3 |   IN | 0 | 11 || 12 | 0 | IN   | PC09     | 6   | 73   |
 |  119 |   7 |    TXD.3 |   IN | 0 | 13 || 14 |   |      | GND      |     |      |
 |  122 |   8 |    CTS.3 |   IN | 0 | 15 || 16 | 0 | IN   | PC08     | 9   | 72   |
 |      |     |     3.3V |      |   | 17 || 18 | 0 | IN   | PC07     | 10  | 71   |
 |   66 |  11 |   MOSI.0 |   IN | 0 | 19 || 20 |   |      | GND      |     |      |
 |   67 |  12 |   MISO.0 |   IN | 0 | 21 || 22 | 0 | IN   | RTS.3    | 13  | 121  |
 |   64 |  14 |   SCLK.0 |   IN | 0 | 23 || 24 | 0 | IN   | CE.0     | 15  | 69   |
 |      |     |      GND |      |   | 25 || 26 | 0 | IN   | PH03     | 16  | 227  |
 +------+-----+----------+------+---+----++----+---+------+----------+-----+------+
 | GPIO | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | GPIO |
 +------+-----+----------+------+---+  OPi H6  +---+------+----------+-----+------+
```

orangepi-xunlong, 2020-06-22:
Sorry, there are some problems with the last updated makefile, [this commit](https://github.com/orangepi-xunlong/wiringOP/commit/5ceb4e599a845da665d00ea2751a90789b4b3908) should have fixed the problem, so you can update the code and try again, thanks.

inactive-virus, 2020-06-22:
Thanks for your quick response. It works now.

closed 2020-06-22 by inactive-virus