orangepi-xunlong/wiringOP #23: Orange Pi Bionic with Linux 5.4.27-sunxi problem with ./build script

AlexeySpiridonov, 2020-07-23:
root@orangepizerolts:~/WiringOP# ./build
wiringPi Build script
=====================

WiringPi Library
[UnInstall]
[Compile] wiringPi.c
[Compile] wiringSerial.c
[Compile] wiringShift.c
[Compile] piHiPri.c
[Compile] piThread.c
[Compile] wiringPiSPI.c
[Compile] wiringPiI2C.c
[Compile] softPwm.c
In file included from wiringPi.c:81:0:
wiringPi.c: In function ‘piBoardId’:
OrangePi.h:140:28: error: ‘MEM_INFO’ undeclared (first use in this function); did you mean ‘EMLINK’?
 #define ORANGEPI_MEM_INFO  MEM_INFO
                            ^
wiringPi.c:945:10: note: in expansion of macro ‘ORANGEPI_MEM_INFO’
   *mem = ORANGEPI_MEM_INFO;
          ^~~~~~~~~~~~~~~~~
OrangePi.h:140:28: note: each undeclared identifier is reported only once for each function it appears in
 #define ORANGEPI_MEM_INFO  MEM_INFO
                            ^
wiringPi.c:945:10: note: in expansion of macro ‘ORANGEPI_MEM_INFO’
   *mem = ORANGEPI_MEM_INFO;
          ^~~~~~~~~~~~~~~~~
wiringPi.c: In function ‘pinModeAlt’:
OrangePi.h:138:31: error: ‘GPIO_NUM’ undeclared (first use in this function); did you mean ‘GPIO_PWM’?
 #define PI_GPIO_MASK       (~(GPIO_NUM - 1))
                               ^
wiringPi.c:1339:14: note: in expansion of macro ‘PI_GPIO_MASK’
   if ((pin & PI_GPIO_MASK) == 0)  // On-board pin
              ^~~~~~~~~~~~
wiringPi.c: In function ‘pinMode’:
OrangePi.h:138:31: error: ‘GPIO_NUM’ undeclared (first use in this function); did you mean ‘GPIO_PWM’?
 #define PI_GPIO_MASK       (~(GPIO_NUM - 1))
                               ^
wiringPi.c:1374:14: note: in expansion of macro ‘PI_GPIO_MASK’
   if ((pin & PI_GPIO_MASK) == 0) {
              ^~~~~~~~~~~~
wiringPi.c: In function ‘pullUpDnControl’:
OrangePi.h:138:31: error: ‘GPIO_NUM’ undeclared (first use in this function); did you mean ‘GPIO_PWM’?
 #define PI_GPIO_MASK       (~(GPIO_NUM - 1))
                               ^
wiringPi.c:1495:14: note: in expansion of macro ‘PI_GPIO_MASK’
   if ((pin & PI_GPIO_MASK) == 0)  // On-Board Pin
              ^~~~~~~~~~~~
wiringPi.c: In function ‘digitalRead’:
OrangePi.h:138:31: error: ‘GPIO_NUM’ undeclared (first use in this function); did you mean ‘GPIO_PWM’?
 #define PI_GPIO_MASK       (~(GPIO_NUM - 1))
                               ^
wiringPi.c:1533:16: note: in expansion of macro ‘PI_GPIO_MASK’
     if ((pin & PI_GPIO_MASK) == 0) {
                ^~~~~~~~~~~~
wiringPi.c: In function ‘digitalWrite’:
OrangePi.h:138:31: error: ‘GPIO_NUM’ undeclared (first use in this function); did you mean ‘GPIO_PWM’?
 #define PI_GPIO_MASK       (~(GPIO_NUM - 1))
                               ^
wiringPi.c:1632:14: note: in expansion of macro ‘PI_GPIO_MASK’
   if ((pin & PI_GPIO_MASK) == 0) {
              ^~~~~~~~~~~~
wiringPi.c: In function ‘pwmWrite’:
OrangePi.h:135:28: error: ‘GPIO_NUM’ undeclared (first use in this function); did you mean ‘GPIO_PWM’?
 #define MAX_PIN_NUM        GPIO_NUM
                            ^
wiringPi.c:1712:15: note: in expansion of macro ‘MAX_PIN_NUM’
     if (pin < MAX_PIN_NUM) // On-Board Pin needto fix me Jim
               ^~~~~~~~~~~
wiringPi.c: In function ‘wiringPiSetup’:
OrangePi.h:139:28: error: ‘GPIOA_BASE’ undeclared (first use in this function); did you mean ‘GPIO_BASE’?
 #define GPIO_BASE          GPIOA_BASE
                            ^
wiringPi.c:2368:86: note: in expansion of macro ‘GPIO_BASE’
  gpio = (uint32_t *)mmap(0, BLOCK_SIZE * 10, PROT_READ | PROT_WRITE, MAP_SHARED, fd, GPIO_BASE);
                                                                                      ^~~~~~~~~
OrangePi.h:141:23: error: ‘GPIO_PWM_OP’ undeclared (first use in this function); did you mean ‘GPIO_PWM’?
 #define GPIO_PWM      GPIO_PWM_OP
                       ^
wiringPi.c:2375:78: note: in expansion of macro ‘GPIO_PWM’
  pwm = (uint32_t *)mmap(0, BLOCK_SIZE, PROT_READ|PROT_WRITE, MAP_SHARED, fd, GPIO_PWM) ;
                                                                              ^~~~~~~~
Makefile:142: recipe for target 'wiringPi.o' failed
make: *** [wiringPi.o] Error 1
make: *** Waiting for unfinished jobs....

AlexeySpiridonov, 2020-07-23:
on last version from GITHub

alexgubanow, 2022-01-17:
Привет @AlexeySpiridonov проблема все ещё актуальна? Можешь попробовать с последней версии?

einhugur, 2022-03-03:
I still get this problem.

I am running Fresh installed Armbian Buster. And I get exactly same errors as the other user gets.

nccwarp9, 2022-07-22:
> This error actually comes if the board is not detected correctly it seems. I had old Orange PI 2. I was able to get it to build by brute forcing the defines in the code to let it be H3 board.

how did you get it to compile ?

closed 2022-12-23 by AlexeySpiridonov

ZiTAL, 2023-07-11:
same here:

https://github.com/orangepi-xunlong/wiringOP/issues/82