orangepi-xunlong/wiringOP #28: Access gpios without root access

HerrMuellerluedenscheid, 2020-11-03:
Hey,
I have tested wiringOP on armbian running on an orange pi zero LTS. I tried to access GPIOs as a non-root user but can't seem to do so.
```
wiringPiSetup: Unable to open /dev/mem or /dev/gpiomem: No such file or directory.
  Aborting your program because if it can not access the GPIO
  hardware then it most certianly won't work
  Try running with sudo?
```
Running user space software with sudo is clearly not what I want.
I tried already adding the group `gpio` and adding the user to that. I also changed permissions of `/dev/mem` from 
```
crw-r----- 1 root kmem 1, 1 Nov  3 16:59 /dev/mem
```
to
```
crw-rw---- 1 root gpio 1, 1 Nov  3 16:59 /dev/mem
```
as I read in some post but that also didn't fix it.
So how can I fix this?

AndreiCherniaev, 2021-03-30:
I suggest @orangepi-xunlong make **/dev/gpio** in their Ubuntu release. At this time I need access to /dev/mem to get GPIO adresses: _open ("/dev/mem", O_RDWR | O_SYNC | O_CLOEXEC)_. This is wrong way because badboy can read ram... But at this time this is only one way to call wiringPiSetup() in my app. My executable app has name **test**; I am in their folder. 
0) sudo userchmod -aG kmem
and reboot
1) sudo setcap cap_sys_rawio,cap_dac_override=ep ./test

3) @orangepi-xunlong you should close /dev/mem after mmap registers, isn't it? So you should add 
```
int wiringPiSetup (void)
...
  initialiseEpoch () ;
  close(fd); //close /dev/mem
  return 0 ;
}
```

MrrEdgar, 2022-10-13:
@HerrMuellerluedenscheid Were you be able to solve this issue, I just get the OrangePi and as well as you I need to run Gpio without root permissions.

Cutaneus, 2022-11-09:
Has anyone been able to solve the problem?

Chambana, 2023-06-25:
Anyone solved this?

AndreiCherniaev, 2023-08-11:
> Anyone solved this? @orangepi-xunlong we need user access to GPIO in wiringPi not sudo on the OrangePi 5b.

@orangepi-xunlong

hidara2000, 2023-08-29:
PS for the moment I'm using the following launch.json in vscode

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Debug-Python",
            "type": "python",
            "request": "launch",
            "stopOnEntry": false,
            "program": "${file}",
            "cwd": "${workspaceFolder}",
            "console": "integratedTerminal",
            "debugOptions": [
                "RedirectOutput",
                "Sudo"
            ]
        },
    ]
}
```

zjmirving, 2024-04-13:
I start qt with root and solve it

zjmirving, 2024-04-13:
https://github.com/orangepi-xunlong/wiringOP/issues/28
sudo setcap cap_sys_rawio,cap_dac_override=ep ./test

Bag132, 2024-09-30:
Having issues with this trying to access gpio pins from a ros 2 node.

mnvl, 2025-08-25:
I can confirm that it works only with root as of now, and needs access not only to /dev/gpio*, but also to /dev/mem. Please fix!