orangepi-xunlong/wiringOP #29: In orange pi one plus change mode on wPi 1 to off will make system break down

tlz4320, 2020-11-25:
I try to use wiringOP in my one plus, but when I set mode of wPi 1 to off, the system will break down.
But the wPi 1 original mode is off. If I set it to out or in will be all right.

basinilya, 2021-02-28:
This happens to me too. This was the reason why I had to enable the watchdog service for /dev/watchdog. Anyway, when I do:

    gpio mode 9 OFF

the board freezes. Nothing is printed to the serial console and I can't send commands to it or via wi-fi. so I'm not sure how to troubleshoot it. My board is Orange Pi Lite with Armbian 21.02.1 Buster with Linux 5.10.12-sunxi

orangepi-xunlong, 2021-03-09:
When GPIO is OFF, it means IO disable. At present, GPIO can only be set as input or output, but cannot be set as OFF.

closed 2021-03-12 by orangepi-xunlong