orangepi-xunlong/wiringOP #32: Build fails on Orange PI Zero 2 (official debian image)

XFer012, 2021-01-23:
Build fails on Orange PI Zero 2 (official debian image, latest updates):

`[Compile] wpiExtensions.c
[Compile] w25q64.c
[Compile] oled.c
[Compile] OrangePi.c
[Link (Dynamic)]
/usr/bin/ld: wpiExtensions.o:(.bss+0x408): multiple definition of `comDat'; drcNet.o:(.bss+0x400): first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:147: libwiringPi.so.2.46] Error 1`

orangepi-xunlong, 2021-01-25:
Hello, there is no problem with our test. Please send us the complete compilation steps，thanks!

FerGT50, 2021-01-29:
Here they are

`
 git clone https://github.com/orangepi-xunlong/wiringOP.git
Cloning into 'wiringOP'...
remote: Enumerating objects: 311, done.
remote: Counting objects: 100% (311/311), done.
remote: Compressing objects: 100% (231/231), done.
remote: Total 552 (delta 228), reused 141 (delta 77), pack-reused 241
Receiving objects: 100% (552/552), 376.14 KiB | 1.41 MiB/s, done.
Resolving deltas: 100% (336/336), done.
root@orangepizero2:~/src# cd wiringOP/
root@orangepizero2:~/src/wiringOP# ./build clean
wiringPi:   [Clean]
DevLib:     [Clean]
gpio:       [Clean]
Examples:   [Clean]
Gertboard:  [Clean]
PiFace:     [Clean]
Quick2Wire: [Clean]
PiGlow:     [Clean]
scrollPhat: [Clean]
Deb:
root@orangepizero2:~/src/wiringOP# ./build
wiringPi Build script
=====================

WiringPi Library
[UnInstall]
[Compile] wiringPi.c
[Compile] wiringSerial.c
[Compile] wiringShift.c
[Compile] piHiPri.c
[Compile] piThread.c
[Compile] wiringPiSPI.c
[Compile] wiringPiI2C.c
[Compile] softPwm.c
[Compile] softTone.c
[Compile] mcp23008.c
[Compile] mcp23016.c
[Compile] mcp23017.c
[Compile] mcp23s08.c
[Compile] mcp23s17.c
[Compile] sr595.c
[Compile] pcf8574.c
[Compile] pcf8591.c
[Compile] mcp3002.c
[Compile] mcp3004.c
[Compile] mcp4802.c
[Compile] mcp3422.c
[Compile] max31855.c
[Compile] max5322.c
[Compile] ads1115.c
[Compile] sn3218.c
[Compile] bmp180.c
[Compile] htu21d.c
[Compile] ds18b20.c
[Compile] rht03.c
[Compile] drcSerial.c
[Compile] drcNet.c
[Compile] pseudoPins.c
[Compile] wpiExtensions.c
pseudoPins.c: In function ‘myAnalogRead’:
pseudoPins.c:50:16: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
   50 |   int *ptr   = (int *)node->data0 ;
      |                ^
pseudoPins.c: In function ‘myAnalogWrite’:
pseudoPins.c:59:16: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
   59 |   int *ptr   = (int *)node->data0 ;
      |                ^
pseudoPins.c: In function ‘pseudoPinsSetup’:
pseudoPins.c:89:17: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
   89 |   node->data0 = (unsigned int)ptr ;
      |                 ^
[Compile] w25q64.c
[Compile] oled.c
[Compile] OrangePi.c
[Link (Dynamic)]
/usr/bin/ld: wpiExtensions.o:(.bss+0x408): multiple definition of `comDat'; drcNet.o:(.bss+0x400): first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:147: libwiringPi.so.2.46] Error 1
`

closed 2021-03-29 by orangepi-xunlong