orangepi-xunlong/wiringOP #34: Many problems with Orange Pi 4 (pwm, writePin)

Caprimores, 2021-01-28:
**My task is to implement a cooling system. For this I use a laptop fan that has PWM control.**
I apologize in advance for my text, I am from Russia and I use a translator.

**My system and parameters.**
`root@orangepi4:~/123# lsb_release -a`
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.1 LTS
Release:        20.04
Codename:       focal
`root@orangepi4:~/123# uname -a`                                                                                                                                                                            Linux orangepi4 5.10.10-rockchip64 #trunk.77 SMP PREEMPT Mon Jan 25 13:52:10 CET 2021 aarch64 aarch64 aarch64 GNU/Linux
`root@orangepi4:~/123# cat /proc/version`
Linux version 5.10.10-rockchip64 (root@beast) (aarch64-linux-gnu-gcc (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 8.3.0, GNU ld (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 2.32.0.20190321) #trunk.77 SMP PREEMPT Mon Jan 25 13:52:10 CET 2021

**wiringPi downloaded and assembled gihub from this section.**
`root@orangepi4:~/123# gpio -v`
gpio version: 2.46
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty
  * Device tree is enabled.
  *--> OrangePi 4

`root@orangepi4:~/123# gpio readall`

[image]

**I will be working with hardware pin 11 (wiringPi pin 5 or GPIO pin 33). It is currently in input mode. My first action was to try to test the work through the console.** 
`gpio mode 5 out`
`gpio readall`

[image]

**As we can see, the desired pin is switched to OUT mode. I give it the value 1 with the command** 
`gpio write 5 1`
`gpio readall`

[image]

**The pin value changed and my fan started working. Disabling also works if set to 0. Now I'm trying to accomplish the same with the simplest C code.**

#include <wiringPi.h>
#include <softPwm.h>
#include <stdio.h>
#include <math.h>
#define GPIO1 33
int main(int argc, char *argv[]) {
	if (wiringPiSetupGpio() < 0) return 1;
	pinMode(GPIO1, OUTPUT);
	digitalWrite(GPIO1, HIGH);
	printf("Set HIGH\n");
}

**This code initializes pin 33 (Since it uses GPIO, which is equivalent to 5 wiringPi Pin). Sets the value to HIGH and displays a message. Before compiling it, I will transfer my pin to the INPUT mode for clarity of the performance of one function.**

[image]

Now let's create test.c I will compile it with the command
 `gcc -o test test.c -lwiringPi -lm -pthread -lrt -lcrypt`
 and execute it with the command 
`sudo ./test`
`gpio readall`

[image]

**As we can see, our pin went into input mode, but the value is not set. I tried initially to write the command without including sudo, it also does not help, tried reinstalling the library, it does not give any result. I also wanted to test the functionality of wiringPi-Python, but got problems with the installation (many errors on both python2 and python3 versions).**

**Another problem, pin 7 (wiringPi pin 2 or GPIO pin 150), as far as I understand, has a hardware PWM pin according to the instructions and I also found an article in which it was used as I planned.** [https://forum.armbian.com/topic/13753-orangepi-4-temperature-controlled-pwm-fan/](url)

[image]

**But when I try to switch the pin to PWM mode, I get an error and I cannot implement this functionality.**
`gpio mode 5 pwm`

[image]

**Accordingly, I cannot implement softPwm either, because the function also does not work like digitalWrite. Could you please fix the library code and add hardware pwm support if possible on my board.**

AndreiCherniaev, 2021-03-26:
Please fix incorrect coment in OrangePi.c, this function is not for 2g only

/*
 * Set GPIO Mode on **OrangePi 2G-IOT**  
 */
int OrangePi_set_gpio_mode(int pin, int mode)
{
    unsigned int regval = 0;
    unsigned int bank   = pin >> 5;
    unsigned int index  = pin - (bank << 5);
    unsigned int phyaddr = 0;
#ifdef CONFIG_ORANGEPI_2G_IOT
	unsigned int base_address = 0;
#elif (defined **CONFIG_ORANGEPI_RK3399** || defined CONFIG_ORANGEPI_4 || defined CONFIG_ORANGEPI_R1PLUS)

orangepi-xunlong, 2021-03-29:
> 
> @orangepi-xunlong Please fix incorrect coment in OrangePi.c, this function is not for 2g only
> 
> /*
> 
>     * Set GPIO Mode on **OrangePi 2G-IOT**
>       */
>       int OrangePi_set_gpio_mode(int pin, int mode)
>       {
>       unsigned int regval = 0;
>       unsigned int bank   = pin >> 5;
>       unsigned int index  = pin - (bank << 5);
>       unsigned int phyaddr = 0;
>       #ifdef CONFIG_ORANGEPI_2G_IOT
>       unsigned int base_address = 0;
>       #elif (defined **CONFIG_ORANGEPI_RK3399** || defined CONFIG_ORANGEPI_4 || defined CONFIG_ORANGEPI_R1PLUS)

Has been corrected, thanks!

orangepi-xunlong, 2021-03-29:
> 
> 
> **My task is to implement a cooling system. For this I use a laptop fan that has PWM control.**
> I apologize in advance for my text, I am from Russia and I use a translator.
> 
> **My system and parameters.**
> `root@orangepi4:~/123# lsb_release -a`
> No LSB modules are available.
> Distributor ID: Ubuntu
> Description: Ubuntu 20.04.1 LTS
> Release: 20.04
> Codename: focal
> `root@orangepi4:~/123# uname -a` Linux orangepi4 5.10.10-rockchip64 #trunk.77 SMP PREEMPT Mon Jan 25 13:52:10 CET 2021 aarch64 aarch64 aarch64 GNU/Linux
> `root@orangepi4:~/123# cat /proc/version`
> Linux version 5.10.10-rockchip64 (root@beast) (aarch64-linux-gnu-gcc (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 8.3.0, GNU ld (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 2.32.0.20190321) #trunk.77 SMP PREEMPT Mon Jan 25 13:52:10 CET 2021
> 
> **wiringPi downloaded and assembled gihub from this section.**
> `root@orangepi4:~/123# gpio -v`
> gpio version: 2.46
> Copyright (c) 2012-2018 Gordon Henderson
> This is free software with ABSOLUTELY NO WARRANTY.
> For details type: gpio -warranty
> 
>     * Device tree is enabled.
>       *--> OrangePi 4
> 
> 
> `root@orangepi4:~/123# gpio readall`
> 
> [image]
> 
> **I will be working with hardware pin 11 (wiringPi pin 5 or GPIO pin 33). It is currently in input mode. My first action was to try to test the work through the console.**
> `gpio mode 5 out`
> `gpio readall`
> 
> [image]
> 
> **As we can see, the desired pin is switched to OUT mode. I give it the value 1 with the command**
> `gpio write 5 1`
> `gpio readall`
> 
> [image]
> 
> **The pin value changed and my fan started working. Disabling also works if set to 0. Now I'm trying to accomplish the same with the simplest C code.**
> 
> #include <wiringPi.h>
> #include <softPwm.h>
> #include <stdio.h>
> #include <math.h>
> #define GPIO1 33
> int main(int argc, char *argv[]) {
> if (wiringPiSetupGpio() < 0) return 1;
> pinMode(GPIO1, OUTPUT);
> digitalWrite(GPIO1, HIGH);
> printf("Set HIGH\n");
> }
> 
> **This code initializes pin 33 (Since it uses GPIO, which is equivalent to 5 wiringPi Pin). Sets the value to HIGH and displays a message. Before compiling it, I will transfer my pin to the INPUT mode for clarity of the performance of one function.**
> 
> [image]
> 
> Now let's create test.c I will compile it with the command
> `gcc -o test test.c -lwiringPi -lm -pthread -lrt -lcrypt`
> and execute it with the command
> `sudo ./test`
> `gpio readall`
> 
> [image]
> 
> **As we can see, our pin went into input mode, but the value is not set. I tried initially to write the command without including sudo, it also does not help, tried reinstalling the library, it does not give any result. I also wanted to test the functionality of wiringPi-Python, but got problems with the installation (many errors on both python2 and python3 versions).**
> 
> **Another problem, pin 7 (wiringPi pin 2 or GPIO pin 150), as far as I understand, has a hardware PWM pin according to the instructions and I also found an article in which it was used as I planned.** [https://forum.armbian.com/topic/13753-orangepi-4-temperature-controlled-pwm-fan/](url)
> 
> **I just checked the instructions from that site and that PWM use case will work on my fan as well. I think orange pi 4 has hardware pwm.**
> 
> [image]
> 
> **But when I try to switch the pin to PWM mode, I get an error and I cannot implement this functionality.**
> `gpio mode 5 pwm`
> 
> [image]
> 
> **Accordingly, I cannot implement softPwm either, because the function also does not work like digitalWrite. Could you please fix the library code and add hardware pwm support if possible on my board.**

Sorry, PWM is currently unavailable！

alexgubanow, 2022-01-17:
Привет @ArthurKoba поддержка шим реализована здесь для одной или двух плат. На остальных работа не гарантирована, при этом softPWM должна работать везде и ее хватит для задачи управления кулером. Я пытался решить эту проблему для opi z2, но там очень много всего нужно переписать, а в итоге шим есть только на паре пинов и мапить пины нельзя. Если помогло, закрой issue пожалуйста.

closed 2022-01-20 by ArthurKoba