orangepi-xunlong/wiringOP #41: gcc can't compile blink example in Orange Pi Zero

Chandler-Kluser, 2021-12-10:
Trying to compile the example `blink.c` to the Orange Pi Zero:
[image]

This is what I get from `gcc`:
```
root@orangepizero:~/blink# gcc main.c -o blink -lwiringPi -lwiringPiDev
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../libwiringPi.so: undefined reference to `crypt'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../libwiringPi.so: undefined reference to `rint'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../libwiringPi.so: undefined reference to `pthread_join'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../libwiringPi.so: undefined reference to `pthread_create'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../libwiringPi.so: undefined reference to `pow'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../libwiringPi.so: undefined reference to `shm_open'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/10/../../../libwiringPi.so: undefined reference to `pthread_cancel'
collect2: error: ld returned 1 exit status
```

alexgubanow, 2022-01-17:
Hi @Chandler-Kluser you are missing linker options, to be precise libraries. Please try add following args to gcc command:
-lcrypto -lrt -lpthread
Please close issue, once fix confirmed.

Chandler-Kluser, 2022-01-18:
Haven't tried this, I have flashed another image, run ```apt update``` & ```apt upgrade``` as root before compiling it.

Problem solved, thanks to @alexgubanow

closed 2022-01-18 by Chandler-Kluser