orangepi-xunlong/wiringOP #72: OP Zero2 H616 SPI SCLK doesn't work if pinMode is OUTPUT in different exe

holla2040, 2023-03-23:
I've been fighting SPI problems all afternoon.

I'm trying to get a MAX31865 working on OP Zero2 and seeing some problems with the SPI SCLK working correctly.  

After reboot, my spimain.c code works fine, SCLK functions correctly. 

If stop it and run blinkmain.c which set pinMode(SCLK,OUTPUT), SCLK toggles fine.

However, running spimain again, SCLK doesn't toggle during SPI reads or writes.

I suspect this has something to do with ALT4 on pin 14/GPIO230/SCLK.1. Maybe the overlay is setting ALT4 on startup, running pinMode(14,OUTPUT) clears ALT4, then wiringPiSPISetupMode isn't setting ALT4 again. Is this possible?

I've attached my code for the 2 C programs for spi and blink, compile with this.
gcc blinkmain.c -o blinkmain -pthread -lwiringPi -lm -lrt -lcrypt
gcc spimain.c -o spimain -pthread -lwiringPi -lm -lrt -lcrypt

root@calb[591]: uname -a
Linux calb 4.9.318-sun50iw9 #1 SMP PREEMPT Sat Jan 7 21:28:13 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux

[code.zip](https://github.com/orangepi-xunlong/wiringOP/files/11046874/code.zip)

holla2040, 2023-03-23:
Did some more research. From H616_User_Manual_V1.0_cleaned.pdf
9.6.4. Register List GPIO base address 0x0300B000, PH_CFG0 offset 0xFC
9.6.5.21. 0x00FC PH Configure Register 0 (Default Value: 0x7777_7777)

devmem2 doesn't work correctly, used original devmem (which is part of busybox)

after reboot, PH6_SELECT (26:24)
root@calb[501]: busybox devmem 0x300b0fc 32
0x44772222
0100 0100 0111 0111 0010 0010 0010 0010
            ^^^ = 100:SPI1_CLK

after pinMode(SCLK,OUTPUT)
root@calb[518]: busybox devmem 0x300b0fc 32
0x41772222
0100 0001 0111 0111 0010 0010 0010 0010
            ^^^ = 001:Output

setting PH_CFG0 back to SPI1_CLK
busybox devmem 0x300b0fc 32 0x44772222

now SPI1_CLK functions again. 

I'll look through the spi code to see PH_CFG0 and PH_CFG1 gets configured during SPI1 initialization. I suspect that handled through overlays and not done at the application level.

baiywt, 2025-03-31:
@holla2040 As per your findings, the pinMode function switches SCLK to GPIO mode (by setting PH_CFG0), which disables the SPI functionality, making it unavailable for use.

closed 2025-04-01 by orangepi-xunlong