orangepi-xunlong/wiringOP #74: Problem to build wirinngOP on Orange Pi R1 plus LTS

rafaellmario, 2023-04-08:
Hi, I'm having problems to build wringPI on Orange Pi R1 Plus RTS, I'm using the OS "Orangepir1plus-lts 2.1.8 ubuntu focal server linux 5.10.44" available on OrangePi website. When I try to execute the build command the following are displayed: 

 WiringPi Library
[UnInstall]
[Compile] wiringPi.c
[Compile] wiringSerial.c
[Compile] wiringShift.c
[Compile] piHiPri.c
[Compile] piThread.c
[Compile] wiringPiSPI.c
[Compile] wiringPiI2C.c
[Compile] softPwm.c
[Compile] softTone.c
[Compile] mcp23008.c
[Compile] mcp23016.c
[Compile] mcp23017.c
[Compile] mcp23s08.c
[Compile] mcp23s17.c
[Compile] sr595.c
[Compile] pcf8574.c
[Compile] pcf8591.c
[Compile] mcp3002.c
[Compile] mcp3004.c
[Compile] mcp4802.c
[Compile] mcp3422.c
[Compile] max31855.c
[Compile] max5322.c
[Compile] ads1115.c
[Compile] sn3218.c
[Compile] bmp180.c
[Compile] htu21d.c
[Compile] ds18b20.c
[Compile] rht03.c
[Compile] drcSerial.c
[Compile] drcNet.c
[Compile] pseudoPins.c
[Compile] wpiExtensions.c
pseudoPins.c: In function ‘myAnalogRead’:
pseudoPins.c:50:16: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
   50 |   int *ptr   = (int *)node->data0 ;
      |                ^
pseudoPins.c: In function ‘myAnalogWrite’:
pseudoPins.c:59:16: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
   59 |   int *ptr   = (int *)node->data0 ;
      |                ^
pseudoPins.c: In function ‘pseudoPinsSetup’:
pseudoPins.c:89:17: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
   89 |   node->data0 = (unsigned int)ptr ;
      |                 ^
[Compile] w25q64.c
[Compile] oled.c
[Compile] OrangePi.c
OrangePi.c: In function ‘readR’:
OrangePi.c:1645:7: warning: suggest braces around empty body in an ‘else’ statement [-Wempty-body]
 1645 |  else ;
      |       ^
OrangePi.c: In function ‘writeR’:
OrangePi.c:1704:7: warning: suggest braces around empty body in an ‘else’ statement [-Wempty-body]
 1704 |  else ;
      |       ^
OrangePi.c: In function ‘OrangePi_get_gpio_mode’:
OrangePi.c:1916:6: warning: suggest braces around empty body in an ‘else’ statement [-Wempty-body]
 1916 |  else;
      |      ^
OrangePi.c: In function ‘OrangePi_set_gpio_mode’:
OrangePi.c:2052:6: warning: suggest braces around empty body in an ‘else’ statement [-Wempty-body]
 2052 |  else;
      |      ^
OrangePi.c: In function ‘OrangePi_set_gpio_pullUpDnControl’:
OrangePi.c:2261:26: error: ‘GPIOL_BASE’ undeclared (first use in this function); did you mean ‘GPIO3_BASE’?
 2261 |   phyaddr = pullOffset + GPIOL_BASE + ((index >> 4) << 2);
      |                          ^~~~~~~~~~
      |                          GPIO3_BASE
OrangePi.c:2261:26: note: each undeclared identifier is reported only once for each function it appears in
OrangePi.c:2264:26: error: ‘GPIO_BASE_MAP’ undeclared (first use in this function); did you mean ‘GPIO_BASE’?
 2264 |   phyaddr = pullOffset + GPIO_BASE_MAP + (bank * 36) + ((index >> 4) << 2);
      |                          ^~~~~~~~~~~~~
      |                          GPIO_BASE
OrangePi.c: In function ‘OrangePi_digitalWrite’:
OrangePi.c:2423:8: warning: suggest braces around empty body in an ‘else’ statement [-Wempty-body]
 2423 |    else;
      |        ^
OrangePi.c: In function ‘OrangePi_digitalRead’:
OrangePi.c:2522:7: warning: suggest braces around empty body in an ‘else’ statement [-Wempty-body]
 2522 |   else;
      |       ^
make: *** [Makefile:158: OrangePi.o] Error 1
make: *** Waiting for unfinished jobs....

EhsanVahab, 2023-06-29:
I know it seems a little bit late but for the other users who get stuck I comment
install v0.1

mizegit, 2023-07-07:
Hi，I add two define macros in wiringPi/OrangePi.h , I don't know the meaning, 
but it can be built and seems works fine. 

````
diff --git a/wiringPi/OrangePi.h b/wiringPi/OrangePi.h
index a3cf411..053af95 100755
--- a/wiringPi/OrangePi.h
+++ b/wiringPi/OrangePi.h
@@ -150,6 +150,9 @@ extern volatile unsigned int *gpio4_base;
 #define GPIO3_BASE                                                     0xff240000
 #define GPIO_NUM                               (0x40)

+#define GPIOL_BASE                              (0x0)
+#define GPIO_BASE_MAP                          (0x0)
+
 #define GPIO_SWPORTA_DR_OFFSET                                 0x00
 #define GPIO_SWPORTA_DDR_OFFSET                        0x04
 #define GPIO_EXT_PORTA_OFFSET                          0x50
--
````

baiywt, 2025-03-31:
This issue has been fixed. 
https://github.com/orangepi-xunlong/wiringOP/commit/19b8b9c5ae0de907353522f3c1ec3bf12629262f

closed 2025-04-01 by orangepi-xunlong