orangepi-xunlong/wiringOP #80: Orange PI 5: pull up/down resistor setup doesn't work

GuglielmiAndrea, 2023-06-08:
Hello,
I'm using wiringOP successfully to handle GPIO of Orange PI 5 but I'm not able to configure pull-down resistor, it seems by default pull-up resistor is enabled for all inputs.
It's not so nice because I should apply a strong pull-down resistor externally to get low level normally draining more current for nothing.
Please can you help me to make it working somehow, or at least to disable pull-up default mode?
Tri state would be enough applying a resistor my side.
Thanks
Andrea

leeboby, 2023-07-04:
If you are using a Linux image released by OrangePi, it comes pre-installed with WiringOP and supports pull-up/down resistors by default.

The user manual contains instructions on how to use WiringOP.

GuglielmiAndrea, 2023-07-10:
Thanks, I will try again, but anyhow, where I can find source code of that implementation? I need to integrate it in a custom C application, strange it is not available. Can you give me the link? the repository?

leeboby, 2023-07-13:
https://github.com/orangepi-xunlong/wiringOP/commit/e6a192f7cc08b8a892cdbe5c07ad89f8ae8a6471

closed 2023-10-25 by orangepi-xunlong