orangepi-xunlong/wiringOP #82: Orange pi 5: Error building

ZiTAL, 2023-07-11:
```
root@opi5:/opt/wiringOP# ./build
wiringPi Build script
=====================

WiringPi Library
[UnInstall]
[Compile] wiringPi.c
[Compile] wiringSerial.c
[Compile] wiringShift.c
[Compile] piHiPri.c
[Compile] piThread.c
[Compile] wiringPiSPI.c
In file included from wiringPi.c:81:
wiringPi.c: In function ‘piBoardId’:
OrangePi.h:197:28: error: ‘MEM_INFO’ undeclared (first use in this function)
  197 | #define ORANGEPI_MEM_INFO  MEM_INFO
      |                            ^~~~~~~~
wiringPi.c:953:24: note: in expansion of macro ‘ORANGEPI_MEM_INFO’
  953 |                 *mem = ORANGEPI_MEM_INFO;
      |                        ^~~~~~~~~~~~~~~~~
OrangePi.h:197:28: note: each undeclared identifier is reported only once for each function it appears in
  197 | #define ORANGEPI_MEM_INFO  MEM_INFO
      |                            ^~~~~~~~
wiringPi.c:953:24: note: in expansion of macro ‘ORANGEPI_MEM_INFO’
  953 |                 *mem = ORANGEPI_MEM_INFO;
      |                        ^~~~~~~~~~~~~~~~~
[Compile] wiringPiI2C.c
[Compile] softPwm.c
wiringPi.c: In function ‘pinModeAlt’:
OrangePi.h:195:31: error: ‘GPIO_NUM’ undeclared (first use in this function); did you mean ‘GPIO_PWM’?
  195 | #define PI_GPIO_MASK       (~(GPIO_NUM - 1))
      |                               ^~~~~~~~
wiringPi.c:1353:28: note: in expansion of macro ‘PI_GPIO_MASK’
 1353 |                 if ((pin & PI_GPIO_MASK) == 0) {
      |                            ^~~~~~~~~~~~
wiringPi.c: In function ‘pinMode’:
OrangePi.h:195:31: error: ‘GPIO_NUM’ undeclared (first use in this function); did you mean ‘GPIO_PWM’?
  195 | #define PI_GPIO_MASK       (~(GPIO_NUM - 1))
      |                               ^~~~~~~~
wiringPi.c:1417:28: note: in expansion of macro ‘PI_GPIO_MASK’
 1417 |                 if ((pin & PI_GPIO_MASK) == 0) {
      |                            ^~~~~~~~~~~~
wiringPi.c: In function ‘pullUpDnControl’:
OrangePi.h:195:31: error: ‘GPIO_NUM’ undeclared (first use in this function); did you mean ‘GPIO_PWM’?
  195 | #define PI_GPIO_MASK       (~(GPIO_NUM - 1))
      |                               ^~~~~~~~
wiringPi.c:1542:28: note: in expansion of macro ‘PI_GPIO_MASK’
 1542 |                 if ((pin & PI_GPIO_MASK) == 0) {
      |                            ^~~~~~~~~~~~
wiringPi.c: In function ‘digitalRead’:
OrangePi.h:195:31: error: ‘GPIO_NUM’ undeclared (first use in this function); did you mean ‘GPIO_PWM’?
  195 | #define PI_GPIO_MASK       (~(GPIO_NUM - 1))
      |                               ^~~~~~~~
wiringPi.c:1607:30: note: in expansion of macro ‘PI_GPIO_MASK’
 1607 |                   if ((pin & PI_GPIO_MASK) == 0) {
      |                              ^~~~~~~~~~~~
wiringPi.c: In function ‘digitalWrite’:
OrangePi.h:195:31: error: ‘GPIO_NUM’ undeclared (first use in this function); did you mean ‘GPIO_PWM’?
  195 | #define PI_GPIO_MASK       (~(GPIO_NUM - 1))
      |                               ^~~~~~~~
wiringPi.c:1706:28: note: in expansion of macro ‘PI_GPIO_MASK’
 1706 |                 if ((pin & PI_GPIO_MASK) == 0) {
      |                            ^~~~~~~~~~~~
wiringPi.c: In function ‘pwmWrite’:
OrangePi.h:192:28: error: ‘GPIO_NUM’ undeclared (first use in this function); did you mean ‘GPIO_PWM’?
  192 | #define MAX_PIN_NUM        GPIO_NUM
      |                            ^~~~~~~~
wiringPi.c:1786:15: note: in expansion of macro ‘MAX_PIN_NUM’
 1786 |     if (pin < MAX_PIN_NUM) // On-Board Pin needto fix me Jim
      |               ^~~~~~~~~~~
wiringPi.c: In function ‘wiringPiSetup’:
OrangePi.h:196:28: error: ‘GPIOA_BASE’ undeclared (first use in this function); did you mean ‘GPIO_BASE’?
  196 | #define GPIO_BASE          GPIOA_BASE
      |                            ^~~~~~~~~~
wiringPi.c:2444:93: note: in expansion of macro ‘GPIO_BASE’
 2444 |         gpio = (uint32_t *)mmap(0, BLOCK_SIZE * 10, PROT_READ | PROT_WRITE, MAP_SHARED, fd, GPIO_BASE);
      |                                                                                             ^~~~~~~~~
OrangePi.h:198:36: error: ‘GPIO_PWM_OP’ undeclared (first use in this function); did you mean ‘GPIO_PWM’?
  198 | #define GPIO_PWM                   GPIO_PWM_OP
      |                                    ^~~~~~~~~~~
wiringPi.c:2451:85: note: in expansion of macro ‘GPIO_PWM’
 2451 |         pwm = (uint32_t *)mmap(0, BLOCK_SIZE, PROT_READ|PROT_WRITE, MAP_SHARED, fd, GPIO_PWM) ;
      |                                                                                     ^~~~~~~~
make: *** [Makefile:159: wiringPi.o] Error 1
make: *** Waiting for unfinished jobs....
```

ZiTAL, 2023-07-11:
Looking the **orange-pi4-lts** pinout:

[image]

And the **orange-pi5** pinout

[image]

the pinout are the same, but the opposite column, well:

changing the **build** file:

```
        elif [[ -f /etc/armbian-release ]]; then

                source /etc/armbian-release
                [[ $BOARD == orangepi-r1 ]] && BOARD=orangepir1
                [[ $BOARD == orangepi-rk3399 ]] && BOARD=orangepirk3399
                [[ $BOARD == orangepizeroplus2-h3 ]] && BOARD=orangepizeroplus2h3
                [[ $BOARD == orangepizeroplus2-h5 ]] && BOARD=orangepizeroplus2h5

                BOARD="orangepi4-lts"
                export BOARD="orangepi4-lts"
```

```
./build clean
./build
```

it compiles ok

```
root@opi5:/opt/wiringOP# gpio readall
 +------+-----+----------+------+---+OPi 4 LTS +---+---+--+----------+-----+------+
 | GPIO | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | GPIO |
 +------+-----+----------+------+---+----++----+---+------+----------+-----+------+
 |      |     |     3.3V |      |   |  1 || 2  |   |      | 5V       |     |      |
 |   64 |   0 | I2C8_SDA | ALT4 | 1 |  3 || 4  |   |      | 5V       |     |      |
 |   65 |   1 | I2C8_SCL | ALT4 | 1 |  5 || 6  |   |      | GND      |     |      |
 |  150 |   2 |     PWM1 | ALT4 | 1 |  7 || 8  | 1 | ALT4 | I2C3_SCL | 3   | 145  |
 |      |     |      GND |      |   |  9 || 10 | 1 | ALT4 | I2C3_SDA | 4   | 144  |
 |   33 |   5 | GPIO1_A1 | ALT4 | 1 | 11 || 12 | 1 | ALT4 | GPIO1_C2 | 6   | 50   |
 |   35 |   7 | GPIO1_A3 | ALT4 | 1 | 13 || 14 |   |      | GND      |     |      |
 |   92 |   8 | GPIO2_D4 | ALT4 | 1 | 15 || 16 | 1 | ALT4 | GPIO1_C6 | 9   | 54   |
 |      |     |     3.3V |      |   | 17 || 18 | 1 | ALT4 | GPIO1_C7 | 10  | 55   |
 |   40 |  11 | SPI1_TXD | ALT4 | 1 | 19 || 20 |   |      | GND      |     |      |
 |   39 |  12 | SPI1_RXD | ALT4 | 1 | 21 || 22 | 1 | ALT4 | GPIO1_D0 | 13  | 56   |
 |   41 |  14 | SPI1_CLK | ALT4 | 1 | 23 || 24 | 1 | ALT4 | SPI1_CS  | 15  | 42   |
 |      |     |      GND |      |   | 25 || 26 | 1 | ALT4 | GPIO4_C5 | 16  | 149  |
 +------+-----+----------+------+---+----++----+---+------+----------+-----+------+
 | GPIO | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | GPIO |
 +------+-----+----------+------+---+OPi 4 LTS +---+---+--+----------+-----+------+
```

officially this is the **orangepi4-lts** pinout, and my **orange-pi5** has connected in the:

**GPIOD2_D4** the **VIN** of the fan and in **GND** the **GND** of the fan.

Well, the connection is 13, but the opposite way is 12:

```
gpio export 12 out
gpio write 12 1
```
fan is not wortking...

```
gpio export 13 out
gpio write 13 1
```

and this wall neither...

I will try again in another way

leeboby, 2023-07-13:
Please use next branch.

https://github.com/orangepi-xunlong/wiringOP/tree/next

closed 2023-07-13 by ZiTAL