orangepi-xunlong/wiringOP #95: wiringPiSetup: mmap (PWM) failed: Operation not permitted

drozdi70, 2023-11-25:
After installation on orage pi5 plus board:

root@orangepi5-plus:~/work/wiringOP# gpio -v
gpio version: 2.46
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

  * Device tree is enabled.
  *--> Orange Pi 5 Plus
root@orangepi5-plus:~/work/wiringOP# gpio readall
wiringPiSetup: mmap (PWM) failed: Operation not permitted

OS info:
root@orangepi5-plus:~# lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Armbian 23.8.3 bookworm
Release:        12
Codename:       bookworm

chudo9991, 2023-11-25:
> during installation warrning: [Install] chown: warning: '.' should be ':': ‘root.root’
> 
> All Done.

Can you please explain in more detail how to fix this?

drozdi70, 2023-11-25:
the issue is not fixed :(
I gave only the warning I got during installation - maybe it is related to the issue or not...

chudo9991, 2023-11-27:
> the issue is not fixed :( I gave only the warning I got during installation - maybe it is related to the issue or not...

I found this repo with old fork - https://github.com/zhaolei/WiringOP.git

On my OrangePi Lite it works fine

drozdi70, 2023-11-27:
no warnings this time but:
orangepi@orangepi5-plus:~/work/WiringOP$ gpio -v
gpio version: 2.20
Copyright (c) 2012-2014 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

piBoardRev: Unable to determine board revision from /proc/cpuinfo
 -> No "Hardware" line
 ->  You may want to check:
 ->  http://www.lemaker.org/

orangepi@orangepi5-plus:~/work/WiringOP$ gpio readall
piBoardRev: Unable to determine board revision from /proc/cpuinfo
 -> No "Hardware" line
 ->  You may want to check:
 ->  http://www.lemaker.org/

old version not supporting Orange pi 5 Plus :(

jmaks, 2024-01-08:
https://github.com/orangepi-xunlong/wiringOP/issues/96
near-by issue is says about master branch that works.

Try it.
Works for me on opi pc2 board after successfull build.
Also,for board opi5plus you may use python3 modules (http://www.orangepi.org/orangepiwiki/index.php/Orange_Pi_5_Plus#How_to_install_wiringOP-Python)

```
                                                                                                                      
root@opipc2:/opt/wiringOP-master# cd ..                                                                             
root@opipc2:/opt# gpio -v                                                                                           
gpio version: 2.46                                                                                                    
Copyright (c) 2012-2018 Gordon Henderson                                                                              
This is free software with ABSOLUTELY NO WARRANTY.                                                                    
For details type: gpio -warranty                                                                                      
                                                                                                                      
  * Device tree is enabled.                                                                                           
  *--> Xunlong Orange Pi PC 2                                                                                         
root@opipc2:/opt# gpio readall                                                                                      
 +------+-----+----------+------+---+  OPi PC2 +---+------+----------+-----+------+
 | GPIO | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | GPIO |
 +------+-----+----------+------+---+----++----+---+------+----------+-----+------+
 |      |     |     3.3V |      |   |  1 || 2  |   |      | 5V       |     |      |
 |   12 |   0 |    SDA.0 |  OFF | 0 |  3 || 4  |   |      | 5V       |     |      |
 |   11 |   1 |    SCL.0 |  OFF | 0 |  5 || 6  |   |      | GND      |     |      |
 |    6 |   2 |    PWM.1 |  OFF | 0 |  7 || 8  | 0 | OFF  | PC05     | 3   | 69   |
 |      |     |      GND |      |   |  9 || 10 | 0 | OFF  | PC06     | 4   | 70   |
 |    1 |   5 |    RXD.2 |  OFF | 0 | 11 || 12 | 0 | OFF  | PD14     | 6   | 110  |
 |    0 |   7 |    TXD.2 |  OFF | 0 | 13 || 14 |   |      | GND      |     |      |
 |    3 |   8 |    CTS.2 |  OFF | 0 | 15 || 16 | 0 | OFF  | PC04     | 9   | 68   |
 |      |     |     3.3V |      |   | 17 || 18 | 0 | OFF  | PC07     | 10  | 71   |
 |   15 |  11 |   MOSI.1 |  OUT | 1 | 19 || 20 |   |      | GND      |     |      |
 |   16 |  12 |   MISO.1 |  OFF | 0 | 21 || 22 | 0 | OFF  | RTS.2    | 13  | 2    |
 |   14 |  14 |   SCLK.1 |  OFF | 0 | 23 || 24 | 0 | OFF  | CE.1     | 15  | 13   |
 |      |     |      GND |      |   | 25 || 26 | 0 | OFF  | PA21     | 16  | 21   |
 |   19 |  17 |    SDA.1 |  OFF | 0 | 27 || 28 | 0 | OFF  | SCL.1    | 18  | 18   |
 |    7 |  19 |     PA07 |  OFF | 0 | 29 || 30 |   |      | GND      |     |      |
 |    8 |  20 |     PA08 |  OFF | 0 | 31 || 32 | 0 | OFF  | RTS.1    | 21  | 200  |
 |    9 |  22 |     PA09 |  OFF | 0 | 33 || 34 |   |      | GND      |     |      |
 |   10 |  23 |     PA10 |  OFF | 0 | 35 || 36 | 0 | OFF  | CTS.1    | 24  | 201  |
 |  107 |  25 |     PD11 |  OFF | 0 | 37 || 38 | 0 | OFF  | TXD.1    | 26  | 198  |
 |      |     |      GND |      |   | 39 || 40 | 0 | OFF  | RXD.1    | 27  | 199  |
 +------+-----+----------+------+---+----++----+---+------+----------+-----+------+
 | GPIO | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | GPIO |
 +------+-----+----------+------+---+  OPi PC2 +---+------+----------+-----+------+
```

drozdi70, 2024-04-11:
Solution found and working for me on orangepi5:
echo "BOARD=orangepi5plus" | sudo tee /etc/orangepi-release

closed 2024-04-11 by drozdi70

benjajaja, 2025-08-10:
Appeared when switching from armbian kernel 6.1.75 to nixos kernel 6.15.5.
Fixed by adding kernel param `iomem=relaxed`.