orangepi-xunlong/wiringOP #96: wiringPiSetup: mmap (PWM) failed: Invalid argument

thedmdim, 2023-12-10:
Hello!

- orange pi zero h2+ 512mb
- Armbian_23.11.1_Orangepizero_jammy_current_6.1.63

I installed wiringOP like this:
```
git clone https://github.com/orangepi-xunlong/wiringOP.git
cd wiringOP
./build clean
./build
```

When I run `gpio -v` i get:
```
gpio version: 2.46
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

  * Device tree is enabled.
  *--> Xunlong Orange Pi Zero
```

When I run `gpio readall` i get:
```
wiringPiSetup: mmap (PWM) failed: Invalid argument
```

Any suggestion how to get working gpio?

Maybe there is workaround? Actually I only need kinda toogle, so for example I press a button and there runned script.

bobk1955, 2023-12-11:
Yes, I am seeing the same message on an Orange Pi PC2. It seems something changed in an update just a few weeks ago. 
I downloaded and compiled a prior release, and it seems to work, at least on the H5 based PC2 model.

  https://github.com/orangepi-xunlong/wiringOP/archive/refs/heads/master.zip

nategoossen, 2023-12-15:
Same issue. Thought it was something with trying to install it on Armbian 23.11.1 Bookworm but building off that master release works fine. Thanks for the direction.

drozdi70, 2024-01-10:
while building from master.zip:
OrangePi.h:198:36: error: ‘GPIO_PWM_OP’ undeclared (first use in this function); did you mean ‘GPIO_PWM’?
  198 | #define GPIO_PWM                   GPIO_PWM_OP
      |                                    ^~~~~~~~~~~
wiringPi.c:2451:85: note: in expansion of macro ‘GPIO_PWM’
 2451 |         pwm = (uint32_t *)mmap(0, BLOCK_SIZE, PROT_READ|PROT_WRITE, MAP_SHARED, fd, GPIO_PWM) ;
      |                                                                                     ^~~~~~~~
make: *** [Makefile:159: wiringPi.o] Error 1
make: *** Waiting for unfinished jobs....

and the installation fails on my orangepi5 plus

blzhu, 2024-01-14:
> Yes, I am seeing the same message on an Orange Pi PC2. It seems something changed in an update just a few weeks ago. A "next" branch vs "master" branch issue?? I downloaded and compiled the master release, and it seems to work, at least on the H5 based PC2 model.
> 
> https://github.com/orangepi-xunlong/wiringOP/archive/refs/heads/master.zip

I also meet same issue in OrangePi zero , and  use master branch is OK

yuanwb1984, 2024-01-20:
OrangePi zero plus use master branch is OK

UMI64, 2024-03-07:
I change wiringPi\wiringPi.h line 122
[https://github.com/orangepi-xunlong/wiringOP/blob/next/wiringPi/wiringPi.h](url)
`#define H3_PWM_BASE                             (0x01c21400)`
to
`#define H3_PWM_BASE                             (0x01C21000)`
and it fixed

drozdi70, 2024-03-07:
I recompiled and the same:
orangepi@orangepi5-plus:~/work/wiringOP$ sudo gpio readall
wiringPiSetup: mmap (PWM) failed: Operation not permitted

SavooTerrordasPop100, 2024-03-13:
> Sim, estou vendo a mesma mensagem em um Orange Pi PC2. Parece que algo mudou em uma atualização há apenas algumas semanas. Um problema de ramificação "próxima" versus ramificação "mestre"? Baixei e compilei a versão master e parece funcionar, pelo menos no modelo PC2 baseado em H5.
> 
> https://github.com/orangepi-xunlong/wiringOP/archive/refs/heads/master.zip

I tried downloading from this link but it doesn't work. The error appears: "fatal: unable to acess 'https://github.com/orange.pi-xunlong/wiringOP/archive/refs/heads/master.zip/': Could not resolve host: github.com

elvis-hai-le, 2024-04-26:
Just for anyone who wanted the commands to clone the master branch, this worked for me:
`git clone --branch master https://github.com/orangepi-xunlong/wiringOP.git`
`cd wiringOP`
`./build clean`
`./build`

And then of course, `gpio readall` to see pins.

I am using the OrangePi Lite.

youxiangfade, 2024-05-08:
> 仅适用于任何想要命令克隆主分支的人，这对我有用： `git clone --branch master https://github.com/orangepi-xunlong/wiringOP.git` `cd wiringOP` `./build clean` `./build`
> 
> 然后当然，要看别针。`gpio readall`
> 
> 我正在使用 OrangePi Lite。

I followed your approach and successfully solved the problem. Thanks

> Just for anyone who wanted the commands to clone the master branch, this worked for me: `git clone --branch master https://github.com/orangepi-xunlong/wiringOP.git` `cd wiringOP` `./build clean` `./build`
> 
> And then of course, to see pins.`gpio readall`
> 
> I am using the OrangePi Lite.

I followed your approach and successfully solved the problem. Thanks. I am using Orangepi one

WWolf13, 2024-09-20:
> I change wiringPi\wiringPi.h line 122 [https://github.com/orangepi-xunlong/wiringOP/blob/next/wiringPi/wiringPi.h](url) `#define H3_PWM_BASE (0x01c21400)` to `#define H3_PWM_BASE (0x01C21000)` and it fixed

excellent, it works on H5Pi PC2

ZHH-start, 2024-10-31:
这个是因为后面更新把前面已经实现的功能搞出问题了而已——辨别版本更新的方法可能有所不同，而老版本已经无人维护。

可以checkout到老版本当年发布支持的那个提交，这样可以解决问题。如我的板子是lite1，H3芯片：
[image]
可以看到提交是2020年，提交号a846ffd8f5583516cd25c6b08e49fd9cc47728c4。

切换前：
[image]

切换：
[image]

编译：
[image]
[image]

编译完成后：
[image]
可见成功解决问题。

tingyuxuanjin, 2025-01-25:
经过我在orangpi one上的                      最新的版本PWM的基地址是01c21400，与H3芯片的数据手册对应。按照楼上的指导，我将基地址改为01c21000，成功。说明H3数据手册的这个地址是错误的。同时，我回滚到github的历史版本，其中定义的PWM基地址确实是01c21000，可见初始的设置是正确的。回滚历史的方法也可以成功。[image]

baiywt, 2025-03-31:
This issue has been fixed.
https://github.com/orangepi-xunlong/wiringOP/commit/ae776d4eab5191ec787e8f6e055fd5f4da8a49b3

closed 2025-04-01 by orangepi-xunlong