orangepi-xunlong/wiringOP #98: On Orange Pi Zero3 (H618) wiringPiISR() doesn't work

onyx99, 2023-12-17:
gpio: Unable to open GPIO edge interface for pin 9: Permission denied
wiringPiISR: unable to open /sys/class/gpio/gpio79/value: No such file or directory

winitboonkliang, 2024-01-29:
issue same waitfor solution

YVDevelop, 2024-01-30:
how long to wait? I have the same thing here: [Zero 2](https://github.com/orangepi-xunlong/wiringOP/issues/93)
[X3KT0](https://github.com/X3KT0) commented on Nov 18, 2023.  More than two months have already passed... I would like to understand whether I will have time to complete the project?

StephanS8899, 2024-02-17:
Had the same issue with the OrangePi Zero 2.
After stripping down the example for wiringPi Pin 2 (GPIO 73) and issuing `sudo gpio edge 73 falling` for that GPIO pin prior to executing the program the example is working.
Seems that some port init is not complete, at least using Ubuntu 22.04 LTS...

borex99, 2024-03-06:
#101 should fix this error

baiywt, 2025-03-31:
@onyx99 Fixed.  https://github.com/orangepi-xunlong/wiringOP/commit/4fb8cdc2136b417a0489e7529c29699b7fcaeba0

closed 2025-04-01 by orangepi-xunlong