orangepi-xunlong/wiringOP #105: wiringOP not working

PJ5168, 2024-04-14:
Tried to build on Pi PC, 3LTS, 3B.
gpio readall seems not working.
 +------+-----+----------+--------+---+   OPi 3  +---+--------+----------+-----+------+
 | GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
 +------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
 |      |     |     3.3V |        |   |  1 || 2  |   |        | 5V       |     |      |
 |  122 |   0 |    SDA.0 |   ALT2 | 0 |  3 || 4  |   |        | 5V       |     |      |
 |  121 |   1 |    SCL.0 |   ALT2 | 0 |  5 || 6  |   |        | GND      |     |      |
 |  118 |   2 |    PWM.0 |    OFF | 0 |  7 || 8  | 0 | OFF    | PL02     | 3   | 354  |
 +------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
 | GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
 +------+-----+----------+--------+---+   OPi 3  +---+--------+----------+-----+------+

AZQAXH, 2024-04-28:
the same

mikkame, 2024-05-01:
I have same issue

```
gpio version: 2.46
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: ./gpio/gpio -warranty

  * Device tree is enabled.
  *--> OrangePi 3 LTS
```
```
 +------+-----+----------+--------+---+   OPi 3  +---+--------+----------+-----+------+
 | GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
 +------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
 |      |     |     3.3V |        |   |  1 || 2  |   |        | 5V       |     |      |
 |  122 |   0 |    SDA.0 |    OFF | 0 |  3 || 4  |   |        | 5V       |     |      |
 |  121 |   1 |    SCL.0 |    OFF | 0 |  5 || 6  |   |        | GND      |     |      |
 |  118 |   2 |    PWM.0 |    OFF | 0 |  7 || 8  | 0 | OFF    | PL02     | 3   | 354  |
 +------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
 | GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
 +------+-----+----------+--------+---+   OPi 3  +---+--------+----------+-----+------+```

```
Orange Pi Focal with Linux 5.10.75-sun50iw6
```

kerrik21, 2024-05-11:
> if using tag 0.2 it looks working

Can you post detailed workaround for dummies plz? =)

baiywt, 2025-03-31:
@PJ5168 It works fine on my end. Could you provide more details on your setup?

[image]