orangepi-xunlong/wiringOP #111: The wiringPiISR function does not work!!!!

s-ramezani-t, 2024-06-22:
I am using an Orange Pi 4 LTS and the input lag is very important.
When using the wiringPiISR function, I see the following error message:

gpio: Unable to open GPIO direction interface for pin 10: No such file or directory
wiringPiISR: unable to open /sys/class/gpio/gpio55/value: No such file or directory

Is there a way?

zhaosj98, 2024-08-24:
#118 refer to this issue

czf0613, 2024-08-25:
You can take a look at the #119 PR. It's a simple solution.
Just wait for merge request.

baiywt, 2025-03-31:
> I am using an Orange Pi 4 LTS and the input lag is very important. When using the wiringPiISR function, I see the following error message:
>
> gpio: Unable to open GPIO direction interface for pin 10: [...]

Hi, this has been fixed. https://github.com/orangepi-xunlong/wiringOP/commit/e7912890f451f479d2a6908342fac37891bae691

closed 2025-04-01 by orangepi-xunlong