orangepi-xunlong/wiringOP #114: 在Orangepi4 lts上使用wiringOP的softPwm功能就死机

TrinityLeaves, 2024-07-24:
使用设备：Orangepi 4 lts

系统：Orangepi4-lts_3.0.6_ubuntu_jammy_desktop_xfce_linux5.18.5

Orangepi4-lts_3.0.6_ubuntu_focal_desktop_xfce_linux5.10.43

Orangepi4-lts_3.0.6_debian_bullseye_desktop_xfce_linux5.10.43
（以上三个系统均尝试过都会卡死）
程序：wiringOP的examples里的softPwm程序

原先是用python调用发现会死机，换了好几次系统发现使用example中的softPwm.c也会死机，更换了一次设备（tf卡拔过去的）也是一样的问题，因此怀疑是wiringOP的问题，如果4lts实在不能用的话，就只好考虑其他机型了

TrinityLeaves, 2024-07-24:
使用调试串口获取到的内容是
[  132.341682] Unable to handle kernel NULL pointer dereference at virtu                                                                                             al address 0000000000000002
[  132.342492] Mem abort info:
[  132.342759]   ESR = 0x96000005
[  132.343050]   EC = 0x25: DABT (current EL), IL = 32 bits
[  132.343530]   SET = 0, FnV = 0
[  132.343815]   EA = 0, S1PTW = 0
[  132.344106] Data abort info:
[  132.344375]   ISV = 0, ISS = 0x00000005
[  132.344728]   CM = 0, WnR = 0
[  132.345008] user pgtable: 4k pages, 39-bit VAs, pgdp=00000000395eb000
[  132.345587] [0000000000000002] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[  132.346396] Internal error: Oops: 96000005 [#1] SMP
[  132.346841] Modules linked in: hci_uart xfrm_user xfrm_algo uinput fusb30x(C) sprdwl_ng cfg8021                                                                                             1 sprdbt_tty ip_tables x_tables autofs4 pwm_bl motorcomm
[  132.348185] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.10.43 #3.0.6
[  132.348875] Hardware name: OrangePi 4 LTS (DT)
[  132.349289] pstate: a0000085 (NzCv daIf -PAN -UAO -TCO BTYPE=--)
[  132.349851] pc : rk3x_i2c_irq+0xf4/0x2a0
[  132.350221] lr : rk3x_i2c_irq+0x3c/0x2a0
[  132.350580] sp : ffffffc011eebe50
[  132.350888] x29: ffffffc011eebe50 x28: ffffffc011d952e0
[  132.351386] x27: ffffffc0115e7000 x26: 0000000000000000
[  132.351884] x25: ffffffc0115e7000 x24: ffffffc011d952a0
[  132.352381] x23: ffffff80f5893e30 x22: ffffff800165cc00
[  132.352878] x21: 0000000000000034 x20: 0000000000010208
[  132.353375] x19: ffffff80f5893880 x18: 0000000000000000
[  132.353873] x17: 0000000000000000 x16: 0000000000000000
[  132.354371] x15: 0000000000000000 x14: ffffffc011b11580
[  132.354869] x13: ffffffc0e60ff000 x12: 0000000034d4d91d
[  132.355366] x11: 0000000000000000 x10: ffffffc011c0a048
[  132.355863] x9 : ffffffc010ee03a0 x8 : ffffffc011c0a028
[  132.356360] x7 : 0000000000000004 x6 : 0000000000000004
[  132.356858] x5 : 0000000008a9ab27 x4 : 00000000ffffffbc
[  132.357355] x3 : 0000000000000003 x2 : 0000000000000001
[  132.357852] x1 : 0000000000000040 x0 : 0000000000000000
[  132.358351] Call trace:
[  132.358599]  rk3x_i2c_irq+0xf4/0x2a0
[  132.358941]  __handle_irq_event_percpu+0xd8/0x210
[  132.359376]  handle_irq_event_percpu+0x38/0x90
[  132.359788]  handle_irq_event+0x50/0x94
[  132.360150]  handle_fasteoi_irq+0x9c/0x108
[  132.360533]  generic_handle_irq_desc+0x14/0x20
[  132.360944]  __handle_domain_irq+0xc0/0xc8
[  132.361325]  gic_handle_irq+0xc4/0x160
[  132.361675]  el1_irq+0xc8/0x180
[  132.361979]  arch_local_irq_enable+0xc/0x14
[  132.362369]  cpuidle_enter+0x40/0x58
[  132.362709]  do_idle+0x23c/0x260
[  132.363017]  cpu_startup_entry+0x2c/0x44
[  132.363384]  rest_init+0xd0/0xe4
[  132.363690]  arch_call_rest_init+0x18/0x24
[  132.364070]  start_kernel+0x558/0x590
[  132.364423] Code: b9000001 f942fe60 12800864 0a040294 (79400400)
[  132.364981] ---[ end trace 455908214809ea24 ]---
[  132.365405] Kernel panic - not syncing: Oops: Fatal exception in interrupt
[  132.366033] SMP: stopping secondary CPUs
[  132.366567] CRU:
[  132.366768] 00000000: 00000044 00002201 80000000 00000108 00000007 00007f00 00000000 00000000
[  132.367545] 00000020: 00000044 00002201 80000000 00000108 00000007 00007f00 00000000 00000000
[  132.368322] 00000040: 0000006b 00001301 8000031f 00000108 00000007 00007f00 00000000 00000000
[  132.369098] 00000060: 00000040 00002201 8000031f 00000108 00000007 00007f00 00000000 00000000
[  132.369874] 00000080: 00000064 00001301 80000000 00000108 00000007 00007f00 00000000 00000000
[  132.370650] 000000a0: 0000004b 00001301 80000000 00000108 00000007 00007f00 00000000 00000000
[  132.371427] 000000c0: 00000081 00003701 80f00000 00000100 00000007 00007f00 00000000 00000000
[  132.372202] 000000e0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  132.372978] 00000100: 00000100 00000202 00000140 00006202 00000083 00008580 00008320 00000141
[  132.373754] 00000120: 00000181 00008181 00000141 00000141 00004141 00008340 00003185 00000201
[  132.374530] 00000140: 00000201 00000300 00002e87 0000033f 00009381 00008387 00001103 00005282
[  132.375305] 00000160: 00008247 00000187 00001743 0000001f 000000c6 00000300 00000180 00000002
[  132.376080] 00000180: 0000e300 00008200 00000200 00000200 00000200 00000200 00003f3f 00000081
[... 38 lines cut ...]
[  132.405543] 000000a0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  132.406319] 000000c0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  132.407096] 000000e0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  132.407872] 00000100: 00000dfc 00000180 0000000f 00000000 00000024 00000000 00000000 00000000
[  132.408643] 00000120: 00000000 00000000 00000000 00000000 00000003
[  132.409198] PMU:
[  132.409398] 00000000: 00000000 00000000 00000000 00000000 00000000 40cf0000 40cf0000 00000000
[  132.410175] 00000020: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  132.410951] 00000040: 00000000 00000000 00000000 00000000 00000000 c0cf833f 00000000 00000000
[  132.411729] 00000060: 10000678 10000678 10000678 00000010 00000000 00007e00 00000000 28803038
[  132.412505] 00000080: 000003c0 00011940 00005dc0 000003c0 00005dc0 00000060 0000000f 00000018
[... cut ...]
[  132.417515] 00000000: 00000200 00030133 00000031 00000002 00000000 00000000 00000000 00000000
[  132.418292] 00000020: 00000236 00000237 00000000 00000000 0000024e 00000000 00000000 00000000
[  132.419068] 00000040: 00000279 00000279 00000000 00000000 00000000 00000000 00000000 00000000
[  132.419844] 00000060: 00000004 00000004 00000753 00000753 00000000 00000000 00000000 00000000
[  132.420607] 00000080: 00000000 00000000
[  132.420979] ---[ end Kernel panic - not syncing: Oops: Fatal exception in interrupt ]---

TrinityLeaves, 2024-08-07:
实测3b和5b都不会出现这个问题，应该是4lts的问题了，4lts无法用wiringpi调pwm 开softpwm几下就死机

baiywt, 2025-03-31:
@TrinityLeaves 测试代码能发过来看下吗？我这边测试下看看。