orangepi-xunlong/wiringOP #118: 反馈一处代码错误：wiringPi.c文件中wiringPiISR函数中存在bug，导致wiringpi的isr功能不可用，修复后测试可用

zhaosj98, 2024-08-24:
bug附近的代码
```C
if ((pin < 0) || (pin > 63))
    return wiringPiFailure (WPI_FATAL, "wiringPiISR: pin must be 0-63 (%d)\n", pin) ;

  /**/ if (wiringPiMode == WPI_MODE_UNINITIALISED)
    return wiringPiFailure (WPI_FATAL, "wiringPiISR: wiringPi has not been initialised. Unable to continue.\n") ;
  else if (wiringPiMode == WPI_MODE_PINS)
    bcmGpioPin = pinToGpio [pin] ;
  else if (wiringPiMode == WPI_MODE_PHYS)
    bcmGpioPin = physToGpio [pin] ;
  else
    bcmGpioPin = pin ;

// Now export the pin and set the right edge
//	We're going to use the gpio program to do this, so it assumes
//	a full installation of wiringPi. It's a bit 'clunky', but it
//	is a way that will work when we're running in "Sys" mode, as
//	a non-root user. (without sudo)

  if (mode != INT_EDGE_SETUP)
  {
    /**/ if (mode == INT_EDGE_FALLING)
      modeS = "falling" ;
    else if (mode == INT_EDGE_RISING)
      modeS = "rising" ;
    else
      modeS = "both" ;

    sprintf (pinS, "%d", pin) ;

    if ((pid = fork ()) < 0)	// Fail
      return wiringPiFailure (WPI_FATAL, "wiringPiISR: fork failed: %s\n", strerror (errno)) ;

    if (pid == 0)	// Child, exec
    {
      /**/ if (access ("/usr/local/bin/gpio", X_OK) == 0)
      {
	execl ("/usr/local/bin/gpio", "gpio", "edge", pinS, modeS, (char *)NULL) ;
	return wiringPiFailure (WPI_FATAL, "wiringPiISR: execl failed: %s\n", strerror (errno)) ;
      }
      else if (access ("/usr/bin/gpio", X_OK) == 0)
      {
	execl ("/usr/bin/gpio", "gpio", "edge", pinS, modeS, (char *)NULL) ;
	return wiringPiFailure (WPI_FATAL, "wiringPiISR: execl failed: %s\n", strerror (errno)) ;
      }
      else
	return wiringPiFailure (WPI_FATAL, "wiringPiISR: Can't find gpio program\n") ;
    }
    else		// Parent, wait
      wait (NULL) ;
  }

```

wiringPi.c文件的第4535行`sprintf (pinS, "%d", pin) ;`应该改为`sprintf (pinS, "%d", bcmGpioPin ) ;`，否则会报错：“gpio: Unable to open GPIO direction interface for pin XX: No such file or directory”

我检查了gpio.c文件，根据上面的代码，跟edge有关的部分，使用的管脚编号应该是GPIO编号，而不是输入到wiringPiISR函数的pin

czf0613, 2024-08-24:
能否指教一下具体是哪几行的位置需要修改呢？我也现在发现wiringPiISR不可用

czf0613, 2024-08-25:
我已经改好提到PR里面去了，不知道官方什么时候merge了

baiywt, 2025-03-31:
@czf0613 @zhaosj98 已合并https://github.com/orangepi-xunlong/wiringOP/pull/119

closed 2025-04-01 by orangepi-xunlong