orangepi-xunlong/wiringOP #122: Orange Pi Zero2

mamin27, 2024-10-26:
After installing new Debian:

Welcome to Orange Pi 3.1.0 Bookworm with Linux 6.1.31-sun50iw9

comet@meteo2:~/WiringOP$ gpio -v
gpio version: 2.20
Copyright (c) 2012-2014 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

piBoardRev: Unable to determine board revision from /proc/cpuinfo
 -> No "Hardware" line
 ->  You may want to check:
 ->  http://www.lemaker.org/

baiywt, 2025-03-31:
Hey, can't reproduce this on my end.

[image]

mamin27, 2025-03-31:
After today git pull & build

[image]

orangepi-xunlong, 2025-04-01:
Please reflash the image without modifying anything, then test it and let me know the results. Additionally, please use the wiringOP source code and do not use WiringPi. Thank you.

mamin27, 2025-04-01:
Thanks for your hard work.

[image]

It's working now. You can see that I use I2C,  /dev/i2-c3 and you are showing mode ALT5. What doe's it means?

baiywt, 2025-04-09:
ALT5 means the pin is configured for its 5th alternate function, which in this case is twi3 (I2C3), as shown in the code.

[image]

mamin27, 2025-04-10:
It means that ALT5 -> TWI3 -> SDA.3 (SCK.3)

closed 2025-04-10 by mamin27